ra9fael/blog #146: [fpga] readings

konosubakonoakua, 2025-02-21:
- https://github.com/lastweek/fpga_readings
-

konosubakonoakua, 2025-02-21:
## courses
- [ECE 699 Software/Hardware Codesign Spring 2016](https://people-ece.vse.gmu.edu/coursewebpages/ECE/ECE699_SW_HW/S16/)
- [ECE 6775 High-Level Digital Design Automation Fall 2024](https://www.csl.cornell.edu/courses/ece6775/)
- [MIT Introductory Digital Systems laboratory](https://web.mit.edu/6.111/volume2/www/f2018/index.html)
- []()

konosubakonoakua, 2025-02-21:
## Tutorials
- https://www.realdigital.org/

konosubakonoakua, 2025-02-24:
## Verilog
- https://asic-world.com/verilog/verilog_one_day.html

konosubakonoakua, 2025-02-26:
## simulator
- https://verilator.org/guide/latest/
- https://gtkwave.github.io/gtkwave/

konosubakonoakua, 2025-02-26:
## misc
- https://beaglewire.github.io/

konosubakonoakua, 2025-03-05:
## videos
- [How does a flip flop work, what is metastability and why does it have setup & hold time?](https://www.youtube.com/watch?v=5PRuPVIjEcs)

technomax621, 2025-03-25:
Metastability and Synchronizers

konosubakonoakua, 2025-05-07:
https://tknopp.github.io/RedPitayaDAQServer/dev/index.html

konosubakonoakua, 2025-05-13:
[I/O timing constraints for FPGA/ASIC #4: Source-synchronous output](https://www.linkedin.com/pulse/io-timing-constraints-fpgaasic-4-source-synchronous-output-lukas-vik-fudxc) (article, 2025)
[I/O timing constraints for FPGA/ASIC #3: Sink-synchronous input](https://www.linkedin.com/pulse/io-timing-constraints-fpgaasic-3-sink-synchronous-input-lukas-vik-iuxuf) (article, 2025)
[I/O timing constraints for FPGA/ASIC #2: System-synchronous input](https://www.linkedin.com/pulse/io-timing-constraints-fpgaasic-2-system-synchronous-input-lukas-vik-gpnkf) (article, 2025)
[I/O timing constraints for FPGA/ASIC #1: Source-synchronous input](https://www.linkedin.com/pulse/io-timing-constraints-fpgaasic-1-source-synchronous-input-lukas-vik-0xslf) (article, 2025)
[The many ways to mess up ready/valid handshaking](https://www.linkedin.com/pulse/many-ways-mess-up-readyvalid-handshaking-lukas-vik-2mqdf) (article, 2024)
[Reliable FPGA CDC constraints #5: Asynchronous FIFO](https://www.linkedin.com/pulse/reliable-cdc-constraints-5-asynchronous-fifo-lukas-vik-snlgf) (article, 2024)
[Why is CDC so hard? Presentation at CERN](https://www.youtube.com/watch?v=AA-YJNMIDJw) (conference talk, 2024)
[Simulate FPGA design with Vivado simulation libraries](https://youtu.be/kt4FGFKl5E0) (video tutorial, 2024)
[Reliable FPGA CDC constraints #4: Build tool settings](https://www.linkedin.com/pulse/reliable-cdc-constraints-4-build-tool-settings-lukas-vik-yknsc) (article, 2024)
[Reliable FPGA CDC constraints #3: Pulses](https://www.linkedin.com/pulse/reliable-cdc-constraints-3-pulses-lukas-vik-31tif) (article, 2024)
[Reliable FPGA CDC constraints #2: Counters and FIFOs](https://www.linkedin.com/pulse/reliable-cdc-constraints-2-counters-fifos-lukas-vik-ist5c) (article, 2024)
[Reliable FPGA CDC constraints #1: The simple single-bit level CDC](https://www.linkedin.com/pulse/reliable-cdc-constraints-1-lukas-vik-copcf) (article, 2024)
[Real-time stereoscopic object tracking on FPGA using neural networks](http://liu.diva-portal.org/smash/get/diva2:747961/FULLTEXT01.pdf) (master thesis, 2014)

konosubakonoakua, 2025-06-06:
[philipabbey fpga blog](https://blog.abbey1.org.uk/index.php/technology/fpga/)